veryl-lang/veryl #1372: Copy interface modport variables without inversion of port direction

janschiefer, 2025-03-10:
I can copy the modport ports with inverted direction with ..converse(xxx).

How can I copy modport ports **without** inverting direction?

```
interface test_if2 {    
 var clk      : logic;                                                                  
 var nrst     : logic;                                                                    
 var test_port: logic;                                                                    

 modport master {                                                                             
 clk      : input ,                                                                        
nrst     : input ,                                                                       
 test_port: output,                                                                 
   }                      
                                                             
  modport slave {                                                                           
    clk : input,                                                                         
     nrst: input,                                                                          
    ..converse(master)
      }                

    //This won't work!                                                                   
     modport driver {
          clk: output,                                                                    
        nrst: output,           
          ..master                                                                      
  }

  }
```

dalance, 2025-03-10:
This feature is not added yet because I didn't came up with an appropriate name.
Is there any idea?

janschiefer, 2025-03-10:
Hmm.......
reuse()
copy_ports()
clone()
...

Generally I think this is an very important quality of life feature for the language.

I ran into into it, because I'm trying to implement an AXI4 bus in Veryl, which has a bizarre amount of signals...

dalance, 2025-03-10:
I prefer `clone` in your ideas. `same` may be good because it shows "direction is same".

```systemverilog
modport driver {
    ..clone(master)
}
modport driver {
    ..same(master)
}
```

dalance, 2025-03-10:
Thanks! I'll add this feature.

By the way, are you interesting in adding your AXI4 interface definition to Veryl standard library?
I think if standard interfaces are defined in std, each library using it can be connected easily.

janschiefer, 2025-03-10:
Great!

Yes, I can submit my work on the AXI4 bus once I'm done.

taichi-ishitani, 2025-03-10:
I also define `monitor` modports where all ports are inputs.
https://github.com/rggen/rggen-veryl-rtl/blob/706b2b95c7f4b5f9bc2af27b0d492a638d836cfe/rggen_apb_if.veryl#L42-L53

For this kind of usecase, I think a similar feature overriding the port directions is useful.

janschiefer, 2025-03-10:
Why not use the Rggen-veryl module for interface definitions?

For what I've seen, there is an incredibly comprehensive codebase.

taichi-ishitani, 2025-03-11:
@janschiefer ,

You mean modport defnitions?

* https://github.com/rggen/rggen-veryl-rtl/blob/706b2b95c7f4b5f9bc2af27b0d492a638d836cfe/rggen_apb_if.veryl#L16
* https://github.com/rggen/rggen-veryl-rtl/blob/706b2b95c7f4b5f9bc2af27b0d492a638d836cfe/rggen_apb_if.veryl#L29
* https://github.com/rggen/rggen-veryl-rtl/blob/706b2b95c7f4b5f9bc2af27b0d492a638d836cfe/rggen_apb_if.veryl#L42

dalance, 2025-03-11:
> For this kind of usecase, I think a similar feature overriding the port directions is useful.

`..input` is already added, so the following code can be used.

```systemverilog
modport monitor {
    ..input,
}
```

dalance, 2025-03-11:
I think `rggen_apb_if`, `rggen_avalon_if`, `rggen_axi4lite_if` and `rggen_wishbone_if` is useful for std.

PR #1381 (2025-03-11): Add `same` direction to modport default member
Closes #1372
closed 2025-03-11 by dalance