veryl-lang/veryl #1383: Clock and reset signals inside interfaces not recognized by always_ff

janschiefer, 2025-03-11:
The Veryl parser fails to recognize the clock and reset types as well as their bit width inside interfaces.

```
interface axi_c_r {
    var ACLK   : clock_posedge ;
    var ARESETN: reset_sync_low;

    modport peripherial {
        ACLK   : input,
        ARESETN: input,
    }

}

//Working
module test3 (
    ACLK   : input clock_posedge ,
    ARESETN: input reset_sync_low,

) {

    var test: logic;

    always_ff (ACLK, ARESETN) {
        if_reset {
            test = 'b0;
        } else {
            test = 'b1;
        }
    }
}

/*
ERRORS

1.
invalid_clock (https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#invalid_clock)

  x #axi_ctrl can't be used as a clock because it is not 'clock' type nor a single bit
    | signal

2.
invalid_reset (https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#invalid_reset)

  x #axi_ctrl can't be used as a reset because it is not 'reset' type nor a single bit
    | signal
*/
module test4 (
    axi_ctrl: modport axi_c_r::peripherial,
) {

    var test: logic;

    always_ff (axi_ctrl.ACLK, axi_ctrl.ARESETN) {
        if_reset {
            test = 'b0;
        } else {
            test = 'b1;
        }

    }

}
```
I don't know how difficult this is to implement, but it would be nice if clock and reset signals could tag along with the rest if the signals in an interface

dalance, 2025-03-12:
I'll allow clock and reset inside interface at #1385.
About default clock and reset is kept to be forbidden because it is not clear that whether modport has clock and reset.

```systemverilog
module test4 (
    axi_ctrl: modport axi_c_r::peripherial,
) {
    var test: logic;

    // Error: clock and reset inside modport is not used as default clock and reset
    always_ff {
        test = 'b1;
    }

    // OK
    always_ff (axi_ctrl.ACLK, axi_ctrl.ARESETN) {
        test = 'b1;
    }
}
```

janschiefer, 2025-03-12:
Nice!

PR #1385 (2025-03-12): Support clock and reset inside interface
Closes #1383
closed 2025-03-12 by dalance