veryl-lang/veryl #1390: Interface member function not found when interface is a module port

janschiefer, 2025-03-12:
The Veryl parser fails with "unknown member" when trying to call a function of an Interface when this interface is a module port. It works with an interface instantiated inside the module.

```
axi_if {
  //(...)
  var AWREADY : logic;
  var AWVALID : logic;
  //(...)

  function wdata_ack() -> logic {
  return AWREADY && AWVALID;
  }

  modport master {
  AWREADY: output,
  AWVALID: input
  }

  }

  module test_if5( axi_bus: modport axi_if::master) {
  inst internal_if : axi_if();

  var ack1 : logic;
  var ack2 : logic;

  //Working
  assign ack1 = internal_if.wdata_ack();      

  /* Error: unknown_member (https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#unknown_member)
  x "axi_bus" doesn't have member "wdata_ack"
*/
assign ack2 = axi_bus.wdata_ack();

  }

```

taichi-ishitani, 2025-03-12:
Do you add the function to the modport `master`?
Example:
https://github.com/rggen/rggen-veryl-rtl/blob/master/rggen_bus_if.veryl#L21-L31

janschiefer, 2025-03-12:
Ah. I overlooked the "import" keyword.

closed 2025-03-12 by janschiefer