veryl-lang/veryl #1399: Introduce `connect` operator

taichi-ishitani, 2025-03-14:
refs #1391

The `connect` operator `<>` is to connect two interfaces.

```systemverilog
always_comb {
  master_if <> slave_if;
}
```

The above code is expanded into each port connections like below.

```systemverilog
always_comb begin
  slave_if.ready  = master_if.ready;
  master_if.valid = slave_if.valid;
  master_if.data  = slave_if.data;
end
```

Modport is required to determine directions if an operand is an interface instance.

```systemverilog
always_comb {
 // if_a is an interface instance
 // slave_if is a modport
  if_a.master <> slave_if;
}
```

`connect` keyword is required if a `connect` operation is used in a module body but not an `always_*` procedure.

```systemverilog
module ModuleA {
  connect if_a.master <> slave_if;
}
```

For this case, modport including `intou` ports and an arrray interface instance/modport can be used as an operand.
Such operations will be emitted like below.

* Including inout ports

```systemverilog
tran (if_a.scl, if_b.scl);
tran (if_a.sda, if_b.sda);
```

* Array interface/modport

```systemverilog
for (genvar __i = 0;__i < 4;__i++) begin : __g_connect_if_a_if_b
  alwasy_comb begin
    slave_if[__i].ready = master_if[__i].ready;
    master_if[__i].valid = slave_if[__i].valid;
    master_if[__i].data  = slave_if[__i].data;
  end
end
```

There are some needs to tie input ports of an interface to a specific value.
(e.g. https://github.com/pezy-computing/pzbcm/blob/985e7147ab42682e62f32604433090b3be5f8482/pzcorebus_dummy/pzcorebus_dummy_slave.sv#L36-L47)
`connect` operation should also support such cases.

```systemverilog
always_comb {
  slave_if <> 0;
}
```

This code will emitted like below.

```systemverilog
always_comb begin
  slave_if.ready = 0;
end
```

taichi-ishitani, 2025-03-14:
To connect array interfaces, array sizes of these interfaces need to be matched.
For this purpose, expression comparator is required.
The comparator compares two expression have the same operands and operator does not evaluate expressions.

taichi-ishitani, 2025-03-14:
For array interface connection, part connection should be supported.

```systemverilog
connect a_if[0+:4].master <> slave_if[4+:4];
```

dalance, 2025-03-14:
> To connect array interfaces, array sizes of these interfaces need to be matched. For this purpose, expression comparator is required. The comparator compares two expression have the same operands and operator does not evaluate expressions.

Is checking equivalence without evaluation possible?
I feel even the simplest case like `A + B == B + A` is difficult to handle it generally...

taichi-ishitani, 2025-03-14:
The comparetor does not check values of A and B.
In this case, it compares operand identifier names and the operator kind.

* `A + B` == `B + A` is true
* `A + B` == `A - B` is false
* `A + B` == `A + C` is false

dalance, 2025-03-14:
I agree the very limited case like 2 operands and 1 operator can be solved.
But I feel that slightly complex case becomes to be difficult drastically.

* literal equivalency: `A + 2 == A + 1 + 1`
* commutativity: `A - B + C - D == C - D + A - B`
* associativity: `(A + B) * C == A * C + B * C`
* transitivity: `B == C  -> A + B == A + C`

I think this complexity is a reason for existence of specialized equivalency checker like Formality.

On the other hand, the current implementation of evaluator can evaluate and show equivalency of all above cases.

taichi-ishitani, 2025-03-18:
It is difficult to evaluate array size and array select if they are provated by parameter or generic parameter.
For the first, the connect operation supports non array instance/modport and single enement of array instance/modport.

taichi-ishitani, 2025-03-22:
Cast operation is required if type of output side signal is a user defined type.
Thefore, such types should be defined in packages.

closed 2025-03-27 by dalance