veryl-lang/veryl #1414: Naming convention in std interface

dalance, 2025-03-21:
The first interface will be added to std by #1410.
So we should think what naming convention in std is reasonable.

The candidates that naming convention is applied are below:

* Interface
* Parameters
* Modports 
* Variables
* Functions

For example, at #1410 

* Interface: `_if` suffiex
* Interface parameters: Upper camel case
* Modports: Snake case
* Variables: Upper case
* Functions: Snake case

I guess the reason that all variables are upper case is caused by AXI specification.
But I think we should decide general naming convention for std and use it instead of each independent specification.

taichi-ishitani, 2025-03-21:
How about following Rust's naming rule?
https://github.com/rust-lang/rfcs/blob/master/text/0430-finalizing-naming-conventions.md

taichi-ishitani, 2025-03-21:
This is not a naming rule but I think we should also consider whether or not common interfaces can include clock/reset signals.

dalance, 2025-03-23:
I forgot naming convention of std is already defined.

https://github.com/veryl-lang/veryl/blob/79d2f658b62df1ae80d56180eccbc5952bd8f249/crates/std/veryl/Veryl.toml#L13-L32

So the additional candidate is `suffix_interface = "if"`.

closed 2025-05-04 by dalance