veryl-lang/veryl #1415: AXI interface version in std

dalance, 2025-03-21:
In #1410, `axi_if` covers AXI3 and AXI4.
How we should treat versions?

* `axi_if` covers all versions through parameter or generics
* Split `axi_if` to `axi3_if` and `axi4_if`, and add `axi5_if` in future

janschiefer, 2025-04-07:
FINALLY I got some time to redo the AXI implementation.

I¨ve implemented the axi interfaces in 2 different ways to choose from --> [](https://github.com/janschiefer/test_axi)

**Variant 1 ( axi_if.veryl && axi_test.veryl ): Implement 2 different axi interfaces for AXI4 and AXI3**
- Pros: 
-- Works with open source tools ( [yosys](https://github.com/YosysHQ/yosys), [sv2v](https://github.com/zachjs/sv2v) )
-- No unused wires
- Cons: Code duplication

**Variant 2 ( axi_if_variant.veryl && axi_test_variant.veryl ): AXI3 & AXI4 uniform implementation using proto package** 
- Pros:
-- More concise, code reusage
- Cons:
-- Incompatible with yosys as well as sv2v
-- Unused wires with width of 1 bit since I see no way to purge them conditionally

Interestingly another bug unmasked:
--> " omit_project_prefix = true " is ignored in interface definitions, but not when the interface is instantiated in a module.

dalance, 2025-04-09:
Thank you for your work!

I think `axi3_if/axi4_if` is better than generic `axi_if` because AXI5, 6 and so on in future may not suite the current definition of `axi_if`.
But if all parameters are specified individually, the definition of AXI components becomes busy.

```systemverilog
module axi4_slice::<ADDR_W: const, DATA_W_BYTES: const, ID_W: const, AWUSER_W: const, WUSER_W: const, BUSER_W: const, ARUSER_W: const, RUSER_W: const> (
    slv_if: modport axi4_if::<ADDR_W, DATA_W_BYTES, ID_W, AWUSER_W, WUSER_W, BUSER_W, ARUSER_W, RUSER_W>::slave,
    mst_if: modport axi4_if::<ADDR_W, DATA_W_BYTES, ID_W, AWUSER_W, WUSER_W, BUSER_W, ARUSER_W, RUSER_W>::master,
) {}
```

So from this viewpoint, using proto package may be better like below:

```systemverilog
// define generic component
module axi4_slice::<PKG: axi4_prototype> (
    slv_if: modport axi4_if::<PKG>::slave,
    mst_if: modport axi4_if::<PKG>::master,
) {}

// use the component with a specific AXI4 parameter
alias package my_axi4_pkg = $sv::axi4_pkg::<32, 4, 1, 8, 8, 8, 8, 8>;

module a {
    inst u_slice: axi4_slice::<my_axi4_pkg>;
}
```

Does proto package cause incompatibility with yosys and sv2s?

dalance, 2025-04-09:
About clock and reset, I think it is better that they are not included in interface.
Especially, clock is affected by phsical restriction in ASIC, and various patterns can be considered like below:

```systemverilog
// all axi_if are fully asynchronous
module A (
    axi_clk: input clock<10>,
    axi_if: modport axi4_if::<...>[10],
) {}

// all axi_if are synchronous
module A (
    axi_clk: input clock,
    axi_if: modport axi4_if::<...>[10],
) {}

// some axi_if come from left side of chip, and others from right side
module A (
    axi_clk_from_leftside : input clock,
    axi_clk_from_rightside: input clock,
    axi_if_from_leftside : modport axi4_if::<...>[10],
    axi_if_from_rightside: modport axi4_if::<...>[10],
) {}
```

janschiefer, 2025-04-18:
I updated my porposals.

--> Removed ACLK and ARESET and in turn also master_interconnect and slave_interconnect modports

Variant 1 - Implement 2 different axi interfaces for AXI4 and AXI3: https://github.com/janschiefer/test_axi/blob/main/src/axi_if.veryl

Variant 2 -  AXI3 & AXI4 uniform implementation using proto package: https://github.com/janschiefer/test_axi/blob/main/src/axi_if_variant.veryl 

I also switched to using the Slang Yosys plugin in my test code, which seems necessary for building Veryl generated SV code with OSS toolchains ( https://github.com/veryl-lang/veryl/issues/1499 )

Pull request time?

dalance, 2025-04-19:
How about the following construct?
This causes some code duplication, but I feel it is more clear how to use.

```systemverilog
pub proto package axi3_prototype {}
pub proto package axi4_prototype {}

pub package axi3_pkg::<ADDR_W: const, DATA_W_BYTES: const, ID_W: const> for axi3_prototype {}
pub package axi4_pkg::<ADDR_W: const, DATA_W_BYTES: const, ID_W: const, AWUSER_W: const, WUSER_W: const, BUSER_W: const, ARUSER_W: const, RUSER_W: const> for axi4_prototype {}

pub interface axi3_if::<PKG: axi3_prototype> {}
pub interface axi4_if::<PKG: axi4_prototype> {}
```

janschiefer, 2025-04-20:
As you wish: [axi_if.veryl](https://github.com/janschiefer/test_axi/blob/main/src/axi_if.veryl)

janschiefer, 2025-04-24:
I have revised the proposal further --> [axi_if.veryl](https://github.com/janschiefer/test_axi/blob/main/src/axi_if.veryl)

1. Changed case of all variabled to lowercase

2. Documentation comments --> Normal comments

3. Added prototypes and interface definitions for AXI4-Lite 

What do you think?

dalance, 2025-04-25:
OK. I think this version can be merged.

taichi-ishitani, 2025-04-25:
How about adding utility functions like this.

```systemverilog
function awaddr_ack() -> logic {
    return awready && awvalid;
]
```

In addition, I think separated modports are useful.

```systemverilog
modport write_master {
  awready: input ,
  awvalid: output,
  ...
  wready : input ,
  wvalid : output,
  ...
  bready : output,
  bvalid : input ,
  ...
}
```

janschiefer, 2025-04-25:
Ok...let's make this the ultimate AXI interface implementation: [axi_if.veryl](https://github.com/janschiefer/test_axi/blob/main/src/axi_if.veryl) 😃 

1. Re-introduced convenience functions *_ack() for all AXI channels and versions. ( These now also work with yosys and the slang frontend 👍 )

2. Added modports for write-only and read-only master and slave devices.

3. Fixed 3 typos i overlooked while changing the case of the variable names

taichi-ishitani, 2025-04-25:
I think separated interface declarations is not good for common module definitions (bus arbiter, FIFO, etc.) because such module definitions also need to be separated par protocol versions.
Therefore, I think one interface declaration should be shared for all protocol versions.

taichi-ishitani, 2025-04-25:
I think AXI3, AXI4 and AXI4 Lite should be merged into one interface.

janschiefer, 2025-04-25:
Also: Is there currently a option inveryl to ignore specific signals depending on parameters? (for example ignore awuser wires in AXI3).

taichi-ishitani, 2025-04-25:
In our interface declaration, unused ports are remained but I think it would not be a problem.
On the other hand, such ports are removed from effective payload because it would consume unnecessary resources.
This article describes how to implement such interface but it is written Japanese.
https://qiita.com/taichi-ishitani/items/ec351cc5d793a4382da0#%E6%8E%A5%E7%B6%9A%E3%81%AE%E7%B0%A1%E7%95%A5%E5%8C%96-%E3%83%93%E3%83%83%E3%83%88%E5%88%97%E3%81%AB%E8%A9%B0%E3%82%81%E3%82%8B%E7%B7%A8

janschiefer, 2025-04-25:
I have unified the (proto) interfaces from AXI4, AXI3 and AXI4-Lite: [axi_if_unified.veryl](https://github.com/janschiefer/test_axi/blob/main/src/axi_if_unified.veryl)

Usage:
```
inst a: axi_if_unified::<axi4_pkg::<32, 4, 1, 8, 8, 8, 8, 8>>;

inst b: axi_if_unified::<axi3_pkg::<32, 4, 1>>;

inst c: axi_if_unified::<axi4_lite_pkg::<32, 4>>;
```

🤙

janschiefer, 2025-04-25:
The old version with separate interfaces is here: [axi_if.veryl](https://github.com/janschiefer/test_axi/blob/221d687f1816ce6cfad123f179183460c9f378cd/src/axi_if.veryl)

Pick your poison. 😄

dalance, 2025-04-25:
For example, if user want to implement AXI4-lite device, how can user do it?

* full compliant for merged interface ( AXI3, AXI4, AXI4-lite )
* AXI4-lite only with document which has "if `axi3_pkg` / `axi4_pkg` is specified, it will be broken"

The former requires too implementation cost, documentation has no guarantee in the latter case.
Instead of it, using `axi4_lite_if` is more simple.

dalance, 2025-04-26:
I think axi3_if, axi4_if and axi4_lite_if is better.
Even if we choose axi_if_unified with axi3/axi4 only, user who want to use AXI4 feature can't express axi3_pkg can't be used except writing it to the document.

taichi-ishitani, 2025-04-28:
We have to separate interface definitions par protocol versions if we want to users who uses AXI4 not to the `wid` signal becaues there is no such langurage feature for now.

janschiefer, 2025-05-01:
I may have hit a [tiny roadblock](https://github.com/veryl-lang/veryl/issues/1573).

While preparing the pull request I found out veryl is OK with...
```
module Module82B (
    axi: modport axi3_if::<axi3_pkg::<32, 4, 8>>::slave,
) {}
```

...but will throw an error when you do exactly the same, but import all the types from veryl_std:
```
Error:   × veryl check failed

Error: invisible_identifier (link)

  × cannot refer indentifier __axi3_if__$std___axi3_pkg__32__4__8 because it is invisible at here
    ╭─[/home/jschiefer/Code/veryl_dev/veryl/testcases/veryl/82_axi_if.veryl:45:18]
 44 │ module Module82B (
 45 │     axi: modport $std::axi3_if::<$std::axi3_pkg::<32, 4, 8>>::slave,
    ·                  ─────────────────────────┬────────────────────────
    ·                                           ╰── Error location
 46 │ ) {}
    ╰────
```

janschiefer, 2025-05-02:
Nice working after te most recent fix.

closed 2025-05-04 by dalance