veryl-lang/veryl #1431: `case`/`switch` inconsistent syntax

hellow554, 2025-03-25:
If you use [case/switch as an expression](https://doc.veryl-lang.org/book/05_language_reference/04_expression/05_case_switch.html) you need to write this:

```veryl
case a {
        0               : 1,
        1               : 2,
        3..=5           : 4,
        10'b00_0000_011x: 5, // matches 6 or 7
        default         : 6,
    };
```

If you want to use [case/switch as a statement](https://doc.veryl-lang.org/book/05_language_reference/05_statement/04_case_switch.html) you need to write this:

```veryl
case a {
            0: b = 1;
            1: b = 2;
            2: {
                   b = 3;
                   b = 3;
                   b = 3;
               }
            default: b = 4;
        }
```

---

Question: is it really necesarry to distinguish the two cases (expression and statement)? Can't you use the expression syntax for everything? I would favor to do so.

perlindgren, 2025-03-25:
In Rust the type of the expression of an assignment is the (zero sized) unit type, not sure if there is any equivalent in Veryl, and how the translation would look like. (If unit types/values are not allowed in Veryl, it would simply be a type error.)

/Per

dalance, 2025-03-25:
`;` at the end of case expression is not a part of expression.
The `;` shows the end of statement.

```systemverilog
// the following `;`s are same meaning
let a: logic = 1;
let b: logic = case a {    };

// case expression can be used without `;`
always_comb {
    if case a {    } {
    }
}
```

On the other hand, I think `;` in case statement may be able to be changed for more consistency.

```systemverilog
case a {
    0: b = 1,
    1: b = 2,
    2: {
           b = 3;
           b = 3;
           b = 3;
       }
    default: b = 4,
}
```

dalance, 2025-03-25:
But the above case introduces statement without `;`.
If so, the following case should be allowed?

```
module ModuleA {
    let a: logic = 1   // single statement without `;`
}
```

In Rust, statement without `;` is allowed to express that the block returns the statement result.
But Veryl and SV can't use such feature.

So, maybe, it is better that all statements have `;` for consistency.