veryl-lang/veryl #1439: Variable is marked as unassigned when using `readmemh`

hellow554, 2025-03-26:
```veryl
module Rom (
    clk          : input  clock    ,
    rst          : input  reset    ,
    i_addr       : input  logic<16>,
    o_data       : output logic<32>,
) {
    var memory: logic<32> [65536];

    initial {
        $readmemh("calc.bin", memory);
    }

    assign o_data = memory[i_addr / 4];
}
```

results in

```
# veryl check                                                                                                                                                                                 147ms
Error:   × veryl check failed

Warning: unassign_variable (link)

  ⚠ memory is unassigned
    ╭─[/home/marcel/projects/fpgastuff/veryl/src/rom.veryl:10:9]
  9 │ ) {
 10 │     var memory: logic<32> [65536];
    ·         ───┬──
    ·            ╰── Error location
 11 │ 
    ╰────
  help: 
```

taichi-ishitani, 2025-03-27:
I'm working on this. 

By the way, currently, any system functions and tasks can be used in Veryl code even if it is not synthesizable. 
(E.g. `$fopen`)
Do we need to support such functions and tasks?

dalance, 2025-03-27:
After fixing this issue, this case should be reported as `invalid_statement` error because assignment in `initial` and `final` is forbidden.
So we should think about #823 to enable this case.

hellow554, 2025-03-27:
> After fixing this issue, this case should be reported as `invalid_statement` error because assignment in `initial` and `final` is forbidden. So we should think about [#823](https://github.com/veryl-lang/veryl/issues/823) to enable this case.

Can you provide an example on how to implement a ROM? @taichi-ishitani uses this to implement a ROM on his risc-v project

dalance, 2025-03-27:
> Can you provide an example on how to implement a ROM? [@taichi-ishitani](https://github.com/taichi-ishitani) uses this to implement a ROM on his risc-v project

Your code using readmemh in initial statement is right.
But it can't be used in ASIC projects, so such feature should be gated by something like attribute.
This is a meaning of #823.

dalance, 2025-03-27:
I'll merge #1443, and close this issue.
The remaining issue continues to #1444 .

closed 2025-03-28 by dalance