veryl-lang/veryl #1445: [Feature] Simlify `if` expression

taichi-ishitani, 2025-03-27:
The if expression is used as alternative for System Verilog's conditional operator so more simlified notation is needed.
For example:

```systemverilog
if a ? b : c

if a ? b : if c ? d : e

if a ? if b ? c : d : e
```

taichi-ishitani, 2025-03-28:
@dalance ,

Should the original notation be left?

dalance, 2025-03-28:
The original notation is always redundant because statement can't be placed in `{}`.
So I think changing to new syntax is better.

```systemverilog
if a { x = y + 1; x + y } else { c }  // This can't be used
```

taichi-ishitani, 2025-03-28:
How notate `else if` branches? Which one is preferred?

```systemverilog
if a ? b :
if c ? d : e;
```

```systemverilog
     if a ? b
else if c ? d : e;
```

dalance, 2025-03-28:
I think the first is better because I think it is diffucult to see that `else` becomes `:` but `else if` is kept.

closed 2025-03-31 by dalance