veryl-lang/veryl #1461: `<>` in `always_comb` causes LSP panic and mdbook-veryl failure

dalance, 2025-03-30:
The following code causes some issues:

*  LSP panic at `analyzer.rs:227:81`
*  mdbook-veryl error: `bus_if1.cmd is unassigned`

```systemverilog
interface InterfaceA {
    var cmd  : logic;
    var ready: logic;

    modport master {
        cmd  : output,
        ready: input ,
    }

    modport slave {
        ..converse(master)
    }
}

module ModuleA (
    mst_if1: modport InterfaceA::master,
    slv_if1: modport InterfaceA::slave ,
) {
    inst bus_if1: InterfaceA;

    always_comb {
        mst_if1 <> bus_if1.slave;
        slv_if1 <> bus_if1.master;
    }
}
```

taichi-ishitani, 2025-03-30:
For 2nd issue, the above example can be re-written like below.

```systemverilog
always_comb {
  mst_if1.cmd   = bus_if1.cmd;
  bus_if1.ready = mst_if1.ready;
  bus_if1.cmd   = slv_if1.cmd;
  slv_if1.ready = bus_if1.ready;
}
```

`bus_if1.cmd` is referred before it is assigned so it is expected error.

taichi-ishitani, 2025-03-30:
Currently, this example is emitted in one `always_comb` block like below.

```systemveriog
always_comb begin
  mst_if1.cmd   = bus_if1.cmd;
  bus_if1.ready = mst_if1.ready;
  bus_if1.cmd   = slv_if1.cmd;
  slv_if1.ready = bus_if1.ready;
end
```

Should this `always_comb` block is separated?

```systemverilog
always_comb begin
  mst_if1.cmd   = bus_if1.cmd;
end
always_comb begin
  bus_if1.ready = mst_if1.ready;
end
always_comb begin
  bus_if1.cmd   = slv_if1.cmd;
end
always_comb begin
  slv_if1.ready = bus_if1.ready;
end
```

dalance, 2025-03-30:
Thanks. I understood it.
Splitting the `always_comb` is OK.

```systemverilog
always_comb {
    mst_if1 <> bus_if1.slave;
}
always_comb {
    slv_if1 <> bus_if1.master;
}
```
So I feel using `<>` in `always_comb` is not good because assignment order should be considered.
How about removing `<>` support in `always_comb`? Is there any use case of it instead of `connect`?

taichi-ishitani, 2025-03-30:
I think connect operation in always_comb is useful to change part of modport like below.

```systemverilog
always_comb begin
  req_master_if <> req_slave_if;
  req_master_if.address = repap_address(req_slave_if.address);
end
```

dalance, 2025-03-30:
OK. If so, this issue needs to be fixed.

taichi-ishitani, 2025-03-30:
For splitting always_comb, I'm not sure we can split always_comb safely if there are other assignments around connect operation.

dalance, 2025-03-30:
I didn't intend automatically splitting by emitter, I only understood that this issue can be resolved by manually splitting.
But I feel it is difficult to write and understand the following code correctly.
This is because we can't see assignment direction by `<>`, and we don't know how `<>` and other assignments interact.

```systemverilog
always_comb begin
  req_master_if <> req_slave_if;
  req_master_if.address = remap_address(req_slave_if.address);
end
```

taichi-ishitani, 2025-03-30:
> This is because we can't see assignment direction by <>, and we don't know how <> and other assignments interact.

I think LSP can help this.
How about showing equivalent expanded code by hover?

For 1st issue, I look into it.

taichi-ishitani, 2025-03-31:
veryl-lsp is also panicked for the code below.

```systemverilog
module ModuleA {
    var _a: logic;
    var _b: logic;
    always_comb {
        _a = _b;
        _b = 0;
    }
}
```

taichi-ishitani, 2025-03-31:
LSP removes symbols of entities which it will be analyze from the symbol table before starting analysis.
https://github.com/veryl-lang/veryl/blob/91e59e7f9261f01ccfc840573a7c6cca9e203b71/crates/languageserver/src/server.rs#L692-L696

But `symbol_table::drop` does not remove entries for those symbols from the `var_ref_list`.
This makes inconsistencies in the symbol table and the analyzer will be panicked due to this.
https://github.com/veryl-lang/veryl/blob/91e59e7f9261f01ccfc840573a7c6cca9e203b71/crates/analyzer/src/symbol_table.rs#L559-L578

PR #1466 (2025-03-31): Remove entries defined in the given file from var_ref_list
fix #1461
closed 2025-03-31 by dalance