veryl-lang/veryl #1462: Inconsistencies in generic package member resolution

nblei, 2025-03-30:
I'm playing with an FFT accelerator:

```SystemVerilog
package fp_types_::<Exponent: const, Mantissa: const> {
    struct fp {
        sign    : logic          ,
        exp     : logic<Exponent>,
        mantissa: logic<Mantissa>,
    }

    struct cfp {
        real: fp,
        imag: fp,
    }
}

module bfly::<Exponent: const, Mantissa: const,
                         fp: type = fp_types_::<Exponent, Mantissa>::fp,
                         cfp: type = fp_types_::<Exponent, Mantissa>::cfp> (
    i_clk: input clock,
    i_rst: input reset,
    i_en : input logic,

    // Complex data inputs
    i_a: input fp_types_::<Exponent, Mantissa>::cfp,
    i_b: input cfp                                 ,
    i_w: input cfp                                 , // Twiddle Factor

    // Complex data outputs
    o_a: output cfp,
    o_b: output cfp,
) {
    // Stage 1: Compute B * W
    inst u_mul_real_real: $sv::DW_fp_mult #(
        sig_width      : Mantissa,
        exp_width      : Exponent,
        ieee_compliance: 0       ,
    ) (
        a: i_a.real,
        b: i_b.real,
    );
}
```
While the `a: i_a.real` line has no problems, the `b: i_b.real` line leads to the following:

```
Error:   × veryl check failed

Error: unknown_member (link)

  × "i_b" doesn't have member "real"
    ╭─[/home/nathan/research/FFTAccelerator/src/bfly.veryl:22:12]
 21 │         a: i_a.real,
 22 │         b: i_b.real,
    ·            ────┬───
    ·                ╰── Error location
 23 │     );
    ╰────
  help:
```
The expected behavior is that `cfp` would bind to `fp_types_::<Exponent, Mantissa>::cfp`, and thus `i_a` and `i_b` would resolve to the same type.  However, Veryl is unable to correctly resolve the type for `i_b`.

taichi-ishitani, 2025-03-30:
https://github.com/veryl-lang/veryl/blob/1051489921a7b0756f4db2e2ac654a4e07c81fb0/crates/analyzer/src/symbol_table.rs#L415

It seems that type expression on LSH of parameter declaration is not resolved.
I think this is a cause of this issue.

taichi-ishitani, 2025-03-31:
> [veryl/crates/analyzer/src/symbol_table.rs](https://github.com/veryl-lang/veryl/blob/1051489921a7b0756f4db2e2ac654a4e07c81fb0/crates/analyzer/src/symbol_table.rs#L415)
> 
> Line 415 in [1051489](/veryl-lang/veryl/commit/1051489921a7b0756f4db2e2ac654a4e07c81fb0)
> 
>  context = self.trace_type_kind(context, &x.r#type.kind)?; 
> It seems that type expression on LSH of parameter declaration is not resolved. I think this is a cause of this issue.

`fp` and `cfp` are generic arguments so the above comment may not be related.

taichi-ishitani, 2025-03-31:
By the way, package generic boundary can be used for this case.

```systemverilog
proto package fp_types_proto_pkg {
  const EXPONENT_WIDTH: u32;
  cosnt MANTISSA_WIDTH: u32;

  struct fp {
    sign    : logic                ,
    exp     : logic<EXPONENT_WIDTH>,
    mantissa: logic<MANTISSA_WIDTH>,
  }

  struct cfp {
    real: fp,
    imag: fp,
  }
}

package fp_types_pkg::<EXPONENT: const, MANTISSA: const> {
  const EXPONENT_WIDTH: u32 = EXPONENT;
  cosnt MANTISSA_WIDTH: u32 = MANTISSA;

  struct fp {
    sign    : logic                ,
    exp     : logic<EXPONENT_WIDTH>,
    mantissa: logic<MANTISSA_WIDTH>,
  }

  struct cfp {
    real: fp,
    imag: fp,
  }
}

module bfly::<FP_PKG: fp_types_proto_pkg> (
    i_clk: input clock,
    i_rst: input reset,
    i_en : input logic,

    // Complex data inputs
    i_a: input FP_PKG::cfp,
    i_b: input FP_PKG::cfp,
    i_w: input FP_PKG::cfp,

    // Complex data outputs
    o_a: output FP_PKG::cfp,
    o_b: output FP_PKG::cfp,
) {
    // Stage 1: Compute B * W
    inst u_mul_real_real: $sv::DW_fp_mult #(
        sig_width      : FP_PKG::EXPONENT_WIDTH,
        exp_width      : FP_PKG::MANTISSA_WIDTH,
        ieee_compliance: 0                     ,
    ) (
        a: i_a.real,
        b: i_b.real,
    );
}

module top {
  alias package fp16_pkg = fp_types_pkg::<5, 10>;
  inst u_bfly: bfly::<fp16_pkg> (
    ...
  );
}
```

Then required parameters and typedefs can be handled together. This is good for maintainability.

nblei, 2025-03-31:
And that makes sense, but it is not obvious what the resolution is.

BTW, the resolution is to simply remove the type alias from the generics list and define it in the main body, while keeping the alias usage in the port list.  Not only is the type checker happy with this, but the generated SystemVerilog has the correct types.  I'm very impressed by all the work you guys have put into this language over the past year.