veryl-lang/veryl #1470: Problems with monomorphic types inside monomorphic interface.

nblei, 2025-03-31:
```systemverilog
interface ready_valid_if::<T: type> {
  var ready: logic;
  var valid: logic;
  var value: T;
  
  modport producer {
    valid: output,
    value: output,
    ready: input,
  }

  modport consumer {
    ..converse(producer)
  }

  modport monitor {
    ..input
  }
}

package fp_types_::<Exponent: const, Mantissa: const> {
    struct fp {
        sign    : logic          ,
        exp     : logic<Exponent>,
        mantissa: logic<Mantissa>,
    }

    struct cfp {
        real: fp,
        imag: fp,
    }

    struct bin_datac {
        a: cfp,
        b: cfp,
    }

    struct bin_dataf {
        a: fp,
        b: fp,
    }

    struct bfly_datac {
        a: cfp,
        b: cfp,
        w: cfp,
    }
}

module dw_mul_wrapper #(
    param Exponent      : u32                  = 8                              ,
    param Mantissa      : u32                  = 7                              ,
    param IeeeCompliance: bit                  = 0                              ,
    param Arch          : fp_types::DwMultArch = fp_types::DwMultArch::Pipelined,
) (
    i_clk  : input clock          ,
    i_rst_n: input reset_async_low,

    consumer: modport ready_valid_if::<fp_types_::<Exponent, Mantissa>::bin_dataf>::consumer,
    producer: modport ready_valid_if::<fp_types_::<Exponent, Mantissa>::bin_dataf>::producer,
) {

    type fp      = fp_types_::<Exponent, Mantissa>::fp     ;
    type bin_data = fp_types_::<Exponent, Mantissa>::bin_dataf;

    inst u_mult: $sv::DW_fp_mult #(
      sig_width: Mantissa,
      exp_width: Exponent,
      ieee_compliance: IeeeCompliance,
      arch: Arch,
    ) (
      a : consumer.value.a
    );

}
```

The `ready_valid_if` has monomorphic parameter `T`, which is monomorphized to `bin_dataf`, which is itself a monomorphized struct type, which contains two fields, `a`, and `b`.  Thus, accessing `consumer.value` should give us a struct, whose fields can be accessed via the dot operator.

However:

```
Error:   × veryl check failed

Error: unknown_member (link)

  × "value" doesn't have member "a"
    ╭─[/home/nathan/research/FFTAccelerator/src/dw_mul_wrapper.veryl:23:11]
 22 │     ) (
 23 │       a : consumer.value.a,
    ·           ────────┬───────
    ·                   ╰── Error location
 24 │     );
    ╰────
  help:
```

Also, if I comment out that line and build the project, it creates the following SV package:

```systemverilog
package FFTAccelerator___fp_types___Exponent__Mantissa;
    typedef struct packed {
        logic         sign    ;
        logic [-1:0] exp     ;
        logic [-1:0] mantissa;
    } fp;

    typedef struct packed {
        fp real;
        fp imag;
    } cfp;

    typedef struct packed {
        cfp a;
        cfp b;
    } bin_datac;

    typedef struct packed {
        fp a;
        fp b;
    } bin_dataf;

    typedef struct packed {
        cfp a;
        cfp b;
        cfp w;
    } bfly_datac;
endpackage
```

The `fp` type inside this package is illegal systemverilog.

nblei, 2025-04-01:
Closed this prematurely.  The issue does seem to exist.

nblei, 2025-04-01:
@dalance Can you reopen this issue?

taichi-ishitani, 2025-04-02:
It seems that  this is the same issue with #1462.
I think a new notation needs to be introduced to constrain members which an argument must have.

```systemverilog
interface ready_valid_if::<T: struct(a, b)> {
}
```

taichi-ishitani, 2025-04-03:
> What about a proto type?

Prototype is also Ok.
Types used as a generic parameter should be derived from the prototype typed for the generic parameter. I'm not sure such strictness realy is needed.

nblei, 2025-04-03:
I am a bit confused as to why this is the suggestion (although I agree that type traits like that would be a good thing, in general).

Here, the `ready_valid_if` should take arbitrary type because the interface is truly type-agnostic.

At the same time, the `dw_mul_wrapper` contains full type information.

The `consumer` modport is to an interface with a fully defined type.  Thus, I think this is a bug in the type resolver.

dalance, 2025-04-03:
I agree this is a bug.
In `ready_valid_if`, any information except `T` is a type is not used, so extra constraint is not necessary.

PR #1492 (2025-04-18): Trace member of generic parameter type
Closes #1470