veryl-lang/veryl #1478: `path` field in `dependencies` section does not work

taichi-ishitani, 2025-04-02:
The `path` field can be used to ovreride a dependency by the local implementation.
https://doc.veryl-lang.org/book/06_development_environment/02_dependencies.html#override-by-local-path

However it seems that this usage doens not work well.

I attached the sample env to reproduce this issue.
[veryl.tar.gz](https://github.com/user-attachments/files/19562691/veryl.tar.gz)

```toml
[project]
name = "test"
version = "0.1.0"

[dependencies]
veryl_sample = {version = "0.15.0", github = "veryl-lang/sample", path = "./sample" }
```

According to the output message, tts remote files is used even if the location of `veryl_sample` is ovewritten by the `path` field.

```console
$ veryl build
[INFO ]       Adding dependency (veryl_sample : https://github.com/veryl-lang/sample @ 0.15.0)
[INFO ]   Processing file (/home/taichi/workspace/temp/veryl/test.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo_pkg.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo_reset_sync.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_handshake/async_handshake.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/binary_enc_dec/binary_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/binary_enc_dec/binary_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/linear_sec_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/linear_sec_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/test_linear_sec.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/counter/counter.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/countones/countones.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/countones/onehot.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/delay/delay.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/edge_detector/edge_detector.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/fifo/fifo.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/fifo/fifo_controller.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_counter.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/lfsr/lfsr_galois.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/pkg/types.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/ram/ram.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/demux.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/mux.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/selector_pkg.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/test_selector.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/synchronizer/synchronizer.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/dependencies/b7032300fe145c759e08640d3dc74db5/src/data_if.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/dependencies/b7032300fe145c759e08640d3dc74db5/src/delay.veryl)
[INFO ]       Output filelist (/home/taichi/workspace/temp/veryl/test.f)
```

dalance, 2025-04-02:
Could you check whether #1479 works fine?

taichi-ishitani, 2025-04-02:
> Could you check whether [#1479](https://github.com/veryl-lang/veryl/pull/1479) works fine?

This fix works fine.

```console
$ ./veryl/target/debug/veryl build
[INFO ]       Adding dependency (veryl_sample : https://github.com/veryl-lang/sample @ 0.15.0)
[INFO ]   Processing file (/home/taichi/workspace/temp/veryl/test.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo_pkg.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_fifo/async_fifo_reset_sync.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/async_handshake/async_handshake.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/binary_enc_dec/binary_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/binary_enc_dec/binary_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/linear_sec_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/linear_sec_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/coding/test_linear_sec.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/counter/counter.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/countones/countones.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/countones/onehot.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/delay/delay.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/edge_detector/edge_detector.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/fifo/fifo.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/fifo/fifo_controller.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_counter.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_decoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/gray/gray_encoder.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/lfsr/lfsr_galois.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/pkg/types.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/ram/ram.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/demux.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/mux.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/selector_pkg.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/selector/test_selector.veryl)
[INFO ]   Processing file (/home/taichi/.cache/veryl/std/65da9082c768d63a811f47ed4a2be686ebc4703d15e6e9a24516ee368fdc1ead/synchronizer/synchronizer.veryl)
[INFO ]   Processing file (/home/taichi/workspace/temp/veryl/sample/src/data_if.veryl)
[INFO ]   Processing file (/home/taichi/workspace/temp/veryl/sample/src/delay.veryl)
[INFO ]       Output filelist (/home/taichi/workspace/temp/veryl/test.f)
```

PR #1479 (2025-04-02): Fix path override of dependency is ignored
Closes #1478
closed 2025-04-02 by dalance