veryl-lang/veryl #1480: Width-less number emittion causes bit width mismatch

dalance, 2025-04-02:
Veryl emits width-less number as a number with an appropriate width.

```systemverilog
// Veryl
'h1

// Generate SV
1'h1
```

But, in enum declaration, it causes bit width mismatch.

```systemverilog
// Veryl
enum EnumA: logic<3> {
    X = 'h0,
    Y = 'h1,
    Z = 'h2,
}

// Generated SV
// All member should be 3bit
typedef enum logic[3-1:0] {
    X = 1'h0,
    Y = 1'h1,
    Z = 2'h2,
} EnumA;
```

So witdh-less number has to be emitted as is, maybe.

taichi-ishitani, 2025-04-02:
For SystemVerilog, width of width-less number litaral is 32 bits so EDA tool may report waning or error for a width-less number litral of which width if over 32bits.
Therefore, I think automatic width calcuration for width-less number literal should not be removed.

----

For the above example, width of each label values should be determined by width of enum.
Therefore, how about checking and emitting width of these literals based on width of enum?

dalance, 2025-04-02:
Thanks. I forgot why width calculation was added.

> For the above example, width of each label values should be determined by width of enum.
> Therefore, how about checking and emitting width of these literals based on width of enum?

How about the case that enum witdh is defined as parameter?

taichi-ishitani, 2025-04-02:
> How about the case that enum witdh is defined as parameter?

Minimum width of enum is determined statically by number of enum variants so we can use it.
By using this minum width, we can emit the enum definition like below.

```systemverilog
typedef enum logic [ENUM_WIDTH-1:0] {
  // literal width '2' is calcurated from number of enum variants.
  X = ENUM_WIDTH'(2'h0),
  Y = ENUM_WIDTH'(2'h1),
  Z = ENUM_WIDTH'(2'h2)
}
```

dalance, 2025-04-02:
Casting by `ENUM_WIDTH` may be sufficient because explicit value can be bigger than minimum width?

```systemverilog
typedef enum logic [ENUM_WIDTH-1:0] {
  X = ENUM_WIDTH'(1'h0),
  Y = ENUM_WIDTH'(1'h1),
  Z = ENUM_WIDTH'(2'h2),
  W = ENUM_WIDTH'(64'hffff_ffff_ffff_ffff),
}
```

taichi-ishitani, 2025-04-03:
> Casting by `ENUM_WIDTH` may be sufficient because explicit value can be bigger than minimum width?

Yes, casting is enough. We don't need to calcurate the minimum width.

taichi-ishitani, 2025-04-11:
Width of enum variant has already been checked. We don't need to add an additional check.
https://github.com/veryl-lang/veryl/blob/8bb9758ec04a088fc475fb06593e20dc67deb794/crates/analyzer/src/handlers/check_enum.rs#L52-L66