veryl-lang/veryl #1499: veryl-std: Yosys chokes on "const AAA: type = BBB" / `localparam type AAA = BBB`

janschiefer, 2025-04-05:
I've been tinkering with the veryl-std and noticed that the open source multi-architecture [Yosys](https://github.com/YosysHQ/yosys) synthesis suite has problems with Veryl definitions like
` const AAA: type = BBB,`
that get translated to the SystemVerilog code 
`localparam type AAA = BBB`  .

These definitions are used extensively in veryl-std, like line 14 in counter.veryl:

`const COUNT: type = logic<WIDTH>,`

There is an [Gihub issue](https://github.com/YosysHQ/yosys/issues/3828) regarding SV type parameter support, however no one knows if and when this is implemented. 

This is a problem in my eyes, since I think the standard library that is shipped with Veryl should support all popular synthesis tools _natively_, especially the most popular free and open source one (even if the solution is less elegant than "localparam type ...").

Maybe this can be handled with typedefs?

Note: There is also another problem in the translated SystemVerliog code from counter.veryl in combination Yosys, that I have been unable to pinpoint to a specific functionality. I will open another isse for that later...

taichi-ishitani, 2025-04-10:
> This is a problem in my eyes, since I think the standard library that is shipped with Veryl should support all popular synthesis tools natively

How about adding generated RTLs to sv-test?
https://github.com/chipsalliance/sv-tests

janschiefer, 2025-04-18:
Yes. --> [sv-tests pull-request 7116](https://github.com/chipsalliance/sv-tests/pull/7116)

While investigating this issue and fiddling with sv-tests, I actually discovered that this (and most other generated constructs from veryl-std) work with Yosys and the external [Slang plugin](https://github.com/povik/yosys-slang). 🥇 

Yay!

closed 2025-04-18 by janschiefer