veryl-lang/veryl #1500: [Feature] Separate a `always_ff` block into two blocks with reset and without reset

taichi-ishitani, 2025-04-06:
There are cases that registers with reset and registers without reset want to be handled in the same `always_ff` block.
(e.g. pair of request valid and its payload)
For this purpose, the `allow(missing_reset_statement)` attribute has been introduced.

```systemverilog
#[allow(missing_reset_statement)]
always_ff {
  if_reset {
    valid = 0;
  } else {
    valid = i_valid;
    data  = i_data;
  }
}
```

Its emitted code is like below and a clock enable logic may be inserted to the registers without reset.
https://www.reddit.com/r/FPGA/comments/1joobci/comment/mlkilk8/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

```systemverilog
always_ff @ (posedge i_clk, negedge i_rst) begin
    if (!i_rst) begin
        valid <= 0;
    end else begin
        valid <= i_valid;
        data  <= i_data;
    end
end
```

A new option to separate a `always_ff` block including both registers with and without reset into two `always_ff` blocks is needed to avoid clock enable logic insertion.
The above `always_ff` block will be emitted like below with this option.

```systemverilog
always_ff @ (posedge i_clk, negedge i_rst) begin
    if (!i_rst) begin
        valid <= 0;
    end else begin
        valid <= i_valid;
    end
end
always_ff @ (posedge i_clk, negedge i_rst) begin
    data  <= i_data;
end
```

dalance, 2025-04-06:
This separation is semantic change about behavior during reset, so I think it should be shown syntactically.
SV and Veryl follows the rule "In the condition which a register is not assigned, the register is not updated", this change becomes an exception of the rule. Even if it is shown explicitly, such exception increases language complexity.
I wonder introducing language complexity in exchange for short code is good.

nblei, 2025-04-06:
Why 

```systemverilog
always_ff @ (posedge i_clk, negedge i_rst) begin
    data  <= i_data;
end
```
and not

```systemverilog
always_ff @ (posedge i_clk) begin
    data  <= i_data;
end
```

The point is that we want `data` to synthesize as DFFs without reset, and `valid` to synthesize as a `dff` with a negative-edge asynchronous reset.

taichi-ishitani, 2025-04-06:
@nblei ,

Simply, this is my mistake. `always_ff` block without reset should be like what you pointed.