veryl-lang/veryl #1508: Directory hierarchy when `target = {type = "directory"}`

dalance, 2025-04-09:
When build target is specified as `{type = "directory", path = "..."}`, all files will be emitted into the directory.
Therefore if some source files have the same name, these files are overwritten.
So the directory hierarchy should be kept in the target directory as the same as source like below:

```toml
target = {type = "directory", path = "target"}
```

```
Veryl.toml
src/a/a.veryl
src/b/b.veryl
target/src/a/a.sv
target/src/b/b.sv
```

To remove unnecessary `src` directory in `target`, explicit `source` can be added.

```toml
source = "src"  # by default `source = "."`
target = {type = "directory", path = "target"}
```

```
Veryl.toml
src/a/a.veryl
src/b/b.veryl
target/a/a.sv
target/b/b.sv
```

dalance, 2025-04-09:
I think `source` can have more complex options, but I don't know they are necessary.

```toml
source = {type = "directory", path = "src"}
source = {type = "glob", path = "src/**/*_x.veryl"}
```

perlindgren, 2025-04-09:
I'm thinking how its done in Rust, where you in the `Cargo.toml` define the (top level) binaries and crates for the project. A module can be a file, or folder.  In this way the code itself dictates the paths to larger extent.

dalance, 2025-04-09:
Rust-like default directory structure which is generated by `veryl new` may be useful.

```toml
[build]
source = "src"
target = {type = "directory", path = "target"}
```

```
Veryl.toml
src/
target/
```

dalance, 2025-04-10:
The issue about default directory structure is followed by  #1515.

About the complex pattern of `source`, it is not clear whether it is necessary.
So I adopt simple `source` at the first step, and will merge #1513.

PR #1513 (2025-04-10): Keep directory hierarchy for `target = {type = "directory"}`
Closes #1508