veryl-lang/veryl #1549: Can't call function via modport

nananapo, 2025-04-22:
version
```sh
$ veryl --version
veryl 0.15.0-nightly (a85e3d5 2025-04-22)
```

code
```veryl
interface InterfaceA {
    var ready: logic  ;
    var valid: logic  ;

    modport master {
        ready: input,
        valid: output,
    }

    function ack () -> logic {
        return ready && valid;
    }
}

module ModuleA(
    a1: modport InterfaceA::master,
) {
    inst a2 : InterfaceA;

    let _v1 : logic = a1.ack();
    let _v2 : logic = a2.ack();
}
```

output
```
Error:   × veryl check failed

Error: unknown_member (https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#unknown_member)

  × "a1" doesn't have member "ack"
    ╭─[/home/kanataso/Documents/nananapo/aaa/src/a.veryl:20:23]
 19 │
 20 │     let _v1 : logic = a1.ack();
    ·                       ───┬──
    ·                          ╰── Error location
 21 │     let _v2 : logic = a2.ack();
    ╰────
  help:

```

taichi-ishitani, 2025-04-22:
You need to add `import` notation like this.

https://github.com/rggen/rggen-veryl-rtl/blob/master/rggen_bus_if.veryl#L30-L30

nananapo, 2025-04-23:
thanks. but it might not be documented.

closed 2025-04-23 by nananapo