veryl-lang/veryl #1573: BUG: "Error: invisible_identifier" when using interface with prototype in module declaration with modport from $std

janschiefer, 2025-05-01:
Just while preparing a pull request for the [AXI interface implementation](https://github.com/veryl-lang/veryl/issues/1415), I stumbled upon a bug involving visibility control when importing from veryl_std.

Proposed axi_if.veryl in veryl_std: [HERE](https://github.com/janschiefer/veryl/blob/master/crates/std/veryl/src/axi_if/axi_if.veryl)

**Test code:**
```
module Module82A () {

    //WORKING
    inst a: $std::axi3_if::<$std::axi3_pkg::<32, 4, 8>>;
    inst b: $std::axi4_if::<$std::axi4_pkg::<32, 4, 8, 8, 8, 8, 8, 8>>;
    inst c: $std::axi4_lite_if::<$std::axi4_lite_pkg::<32, 4>>;

    var r: logic<15>;

    always_comb {
        r[0]  = a.awaddr_ack();
        r[1]  = a.wdata_ack();
        r[2]  = a.bresp_ack();
        r[3]  = a.awaddr_ack();
        r[4]  = a.wdata_ack();
        r[5]  = b.awaddr_ack();
        r[6]  = b.wdata_ack();
        r[7]  = b.bresp_ack();
        r[8]  = b.awaddr_ack();
        r[9]  = b.wdata_ack();
        r[10] = c.awaddr_ack();
        r[11] = c.wdata_ack();
        r[12] = c.bresp_ack();
        r[13] = c.awaddr_ack();
        r[14] = c.wdata_ack();
    }

    inst d: Module82B (
        axi: a,
    );

    inst e: Module82C (
        axi: a,
    );

    inst f: Module82D (
        axi: c,
    );

}

//ERROR: invisible_identifier (link)
module Module82B (
    axi: modport $std::axi3_if::<$std::axi3_pkg::<32, 4, 8>>::slave,
) {}

//ERROR: invisible_identifier (link)
module Module82C (
    axi: modport $std::axi4_if::<$std::axi4_pkg::<32, 4, 8, 8, 8, 8, 8, 8>>::master,
) {}

//ERROR: invisible_identifier (link)
module Module82D (
    axi: modport $std::axi4_lite_if::<$std::axi4_lite_pkg::<32, 4>>::monitor,
) {}
```

**veryl check:**
```
Error:   × veryl check failed

Error: invisible_identifier (link)

  × cannot refer indentifier __axi3_if__$std___axi3_pkg__32__4__8 because it is invisible at here
    ╭─[/home/jschiefer/Code/veryl_dev/veryl/testcases/veryl/82_axi_if.veryl:45:18]
 44 │ module Module82B (
 45 │     axi: modport $std::axi3_if::<$std::axi3_pkg::<32, 4, 8>>::slave,
    ·                  ─────────────────────────┬────────────────────────
    ·                                           ╰── Error location
 46 │ ) {}
    ╰────
  help: 

Error: invisible_identifier (link)

  × cannot refer indentifier __axi4_if__$std___axi4_pkg__32__4__8__8__8__8__8__8 because it is invisible at here
    ╭─[/home/jschiefer/Code/veryl_dev/veryl/testcases/veryl/82_axi_if.veryl:49:18]
 48 │ module Module82C (
 49 │     axi: modport $std::axi4_if::<$std::axi4_pkg::<32, 4, 8, 8, 8, 8, 8, 8>>::master,
    ·                  ─────────────────────────────────┬────────────────────────────────
    ·                                                   ╰── Error location
 50 │ ) {}
    ╰────
```

However, if you put axi_if.veryl and the test file in the **same** project directory and remove the "$std::"-prefix before the types, there is no error.

taichi-ishitani, 2025-05-02:
I'm looking into this.

taichi-ishitani, 2025-05-02:
Mangled identifier `__axi4_if__$std___axi4_pkg__32__4__8__8__8__8__8__8` is invalid identifier for SV because it includes `$` charactor.
It should also be fixed.

closed 2025-05-02 by dalance