veryl-lang/veryl #1578: Interfaces and packages from veryl_std generate inconsistent and invalid names with "$std"

janschiefer, 2025-05-02:
I updated my [commit](https://github.com/veryl-lang/veryl/pull/1577) and split axi*_if and axi*_pkg.

However another problem unmasked.

Inconsistent interface and package names (partially illegal names) are generated in the System Verilog code. 
You only notice them if you run verilator or a synthesis tool (yosys-slang for example here). Veryl check runs without errors.

Verilator:
```
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_pkg.sv:56:9: syntax error, unexpected ${pli-system}, expecting IDENTIFIER or IDENTIFIER-for-cell or IDENTIFIER-for-type or randomize
[DEBUG]    Verilator :    56 | package $std___axi4_pkg__32__4__1__8__8__8__8__8;
[DEBUG]    Verilator :       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator :         ... See the manual at https://verilator.org/verilator_doc.html?v=5.035 for more assistance.
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_pkg.sv:87:1: syntax error, unexpected endpackage
[DEBUG]    Verilator :    87 | endpackage
[DEBUG]    Verilator :       | ^~~~~~~~~~
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:45:11: syntax error, unexpected ${pli-system}, expecting IDENTIFIER or IDENTIFIER-for-cell or IDENTIFIER-for-type or randomize
[DEBUG]    Verilator :    45 | interface $std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8;
[DEBUG]    Verilator :       |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator : %Error-PKGNODECL: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:47:12: Package/class 'std___05F_axi4_pkg___05F32___05F4___05F1___05F8___05F8___05F8___05F8___05F8' not found, and needs to be predeclared (IEEE 1800-2023 26.3)
[DEBUG]    Verilator :    47 |     import std___axi4_pkg__32__4__1__8__8__8__8__8::*;
[DEBUG]    Verilator :       |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator :                   ... For error description see https://verilator.org/warn/PKGNODECL?v=5.035
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:47:12: Importing from missing package 'std___05F_axi4_pkg___05F32___05F4___05F1___05F8___05F8___05F8___05F8___05F8'
[DEBUG]    Verilator :    47 |     import std___axi4_pkg__32__4__1__8__8__8__8__8::*;
[DEBUG]    Verilator :       |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:123:5: syntax error, unexpected modport
[DEBUG]    Verilator :   123 |     modport master (
[DEBUG]    Verilator :       |     ^~~~~~~
[DEBUG]    Verilator : %Error: Exiting due to 6 error(s)
```

yosys-slang:
```
(...)
1. Executing SLANG frontend.
Top level design units:
    top_hw

../output/main.sv:8:5: error: unknown module 'std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8'
    std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8 axi ();
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../output/axi_regfile_memory.sv:9:5: error: unknown interface 'std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8'
    std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8.slave axi
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../output/test_master2.sv:9:5: error: unknown interface 'std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8'
    std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8.master         axi     ,
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../dependencies/std/axi_if/axi_pkg.sv:56:8: error: expected identifier
package $std___axi4_pkg__32__4__1__8__8__8__8__8;
       ^
../dependencies/std/axi_if/axi_if.sv:45:10: error: expected identifier
interface $std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8;
(...)
```

taichi-ishitani, 2025-05-02:
I'v already fixed it.
https://github.com/veryl-lang/veryl/pull/1576

janschiefer, 2025-05-03:
Sorry, I have to reopen this.

The lastest GIT version (03..5.2025) still produces "std"-gibberish:

Yosys-slang:
```
../output/main.sv:8:5: error: unknown module '__std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8'
    __std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8 axi ();
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../output/axi_regfile_memory.sv:9:5: error: unknown interface '__std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8'
    __std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8.slave axi
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../output/test_master2.sv:9:5: error: unknown interface '__std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8'
    __std___axi4_if____std___axi4_pkg__32__4__1__8__8__8__8__8.master         axi     ,
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../dependencies/std/axi_if/axi_pkg.sv:62:8: error: expected identifier
package $std___axi4_pkg__32__4__1__8__8__8__8__8;
       ^
../dependencies/std/axi_if/axi_if.sv:45:10: error: expected identifier
interface $std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8;
         ^
```

Verilator:
```
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_pkg.sv:62:9: syntax error, unexpected ${pli-system}, expecting IDENTIFIER or IDENTIFIER-for-cell or IDENTIFIER-for-type or randomize
[DEBUG]    Verilator :    62 | package $std___axi4_pkg__32__4__1__8__8__8__8__8;
[DEBUG]    Verilator :       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator :         ... See the manual at https://verilator.org/verilator_doc.html?v=5.035 for more assistance.
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_pkg.sv:91:1: syntax error, unexpected endpackage
[DEBUG]    Verilator :    91 | endpackage
[DEBUG]    Verilator :       | ^~~~~~~~~~
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:45:11: syntax error, unexpected ${pli-system}, expecting IDENTIFIER or IDENTIFIER-for-cell or IDENTIFIER-for-type or randomize
[DEBUG]    Verilator :    45 | interface $std___axi4_if__$std___axi4_pkg__32__4__1__8__8__8__8__8;
[DEBUG]    Verilator :       |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator : %Error-PKGNODECL: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:47:12: Package/class '___05Fstd___05F_axi4_pkg___05F32___05F4___05F1___05F8___05F8___05F8___05F8___05F8' not found, and needs to be predeclared (IEEE 1800-2023 26.3)
[DEBUG]    Verilator :    47 |     import __std___axi4_pkg__32__4__1__8__8__8__8__8::*;
[DEBUG]    Verilator :       |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator :                   ... For error description see https://verilator.org/warn/PKGNODECL?v=5.035
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:47:12: Importing from missing package '___05Fstd___05F_axi4_pkg___05F32___05F4___05F1___05F8___05F8___05F8___05F8___05F8'
[DEBUG]    Verilator :    47 |     import __std___axi4_pkg__32__4__1__8__8__8__8__8::*;
[DEBUG]    Verilator :       |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG]    Verilator : %Error: /home/jschiefer/Code/axi_components/dependencies/std/axi_if/axi_if.sv:123:5: syntax error, unexpected modport
[DEBUG]    Verilator :   123 |     modport master (
[DEBUG]    Verilator :       |     ^~~~~~~
[DEBUG]    Verilator : %Error: Exiting due to 6 error(s)
```

taichi-ishitani, 2025-05-03:
I opened PR #1584 for the above issue.

closed 2025-05-03 by dalance