veryl-lang/veryl #1583: number width with _

nananapo, 2025-05-03:
```veryl
module TopModule {
    let _a: u64 = 'b0;
    let _b: u64 = 'b000;
    let _c: u64 = 'b0_00;
}
```
this code is translated into 

```veryl
module project_TopModule;
    longint unsigned _a; always_comb _a = 1'b0;
    longint unsigned _b; always_comb _b = 1'b000;
    longint unsigned _c; always_comb _c = 2'b0_00;
endmodule
```

Is this an intended behavior?

taichi-ishitani, 2025-05-03:
Specified width less than width calucrated from number of digits causes warning reported by simulators or synthesis tools.
Therefore, I think the above examples should be emitted like below.

```systemverilog
_a = 1'b0;
_b = 3'b000;
_c = 3'b0_00;
```

taichi-ishitani, 2025-05-04:
I'm working on it.

closed 2025-05-04 by dalance