veryl-lang/veryl #1608: Shorten mangled name

taichi-ishitani, 2025-05-08:
Mangled name for a generic component with many generic parameters is difficult to read it because it is very long.
For exaxmple:

Veryl
```systemverilog
package pzcorebus_pkg::<
  PROFILE             : pzcorebus_common_pkg::pzcorebus_profile,
  ID_WIDTH            : u16                                    ,
  ADDRESS_WIDTH       : u16                                    ,
  DATA_WIDTH          : u16                                    ,
  UNIT_DATA_WIDTH     : u16                                    ,
  USE_BYTE_ENABLE     : bool                                   ,
  ATOMIC_COMMAND_WIDTH: u16                                    ,
  MESSAGE_CODE_WIDTH  : u16                                    ,
  RQEUEST_INFO_WIDTH  : u16                                    ,
  RESPONSE_INFO_WIDTH : u16                                    ,
  MAX_DATA_WIDTH      : u16                                    ,
  RESPONSE_BOUNDARY   : u16                                    ,
> for pzcorebus_proto_pkg {
```

Generated SV

```systemverilog
package pzbcm___pzcorebus_pkg__pzcorebus_profile_MEMORY_H__16__32__256__32__true__8__8__0__0__256__4096;
```

I think it is useful to shorten such mangled name.

dalance, 2025-05-14:
I agree this concept, but I have some consideration points.

* This should be optional, or by default?
* Index based name is unstable, from the point of view of stability, hash based name may be better? (e.g. `prj___PkgA__a138s9fg`)
* In the generated code, the coment which shows the original parameter or long mangled name may be useful

```
// prj::PkgA::<4, 5, 6, 7>
package prj___PkgA__1;
```

taichi-ishitani, 2025-05-14:
I updated the PR.

* Use hash value instead of instance index
* Print the original instance name as comment

Users can't know the orignal parameters from the shortened generic instace name so I think this feature should be optional.

closed 2025-05-15 by dalance