veryl-lang/veryl #1639: input/output default member emits variable defined in function

nananapo, 2025-05-17:
```veryl
interface IfA {
    var _a: logic;

    function funcA () -> u32 {
        var b: u32;
        b =  0;
        for i: u32 in 0..32 {
            b += i;
        }
        return b;
    }

    modport master{
        ..input
    }
    modport slave{
        ..output
    }
}
```

```sv
interface project_IfA;
    logic _a;

    function automatic int unsigned funcA() ;
        int unsigned b;
        b = 0;
        for (int unsigned i = 0; i < 32; i++) begin
            b += i;
        end
        return b;
    endfunction

    modport master (
        input _a,
        input b 
    );
    modport slave (
        output _a,
        output b 
    );
endinterface
```

taichi-ishitani, 2025-05-18:
I'm working on this.

taichi-ishitani, 2025-05-18:
For the above case, `undefined member` check works well.

closed 2025-05-19 by dalance