veryl-lang/veryl #1673: Anonymous identifiers disallowed in modules with generic parameters

onsdagens, 2025-05-26:
Hi!

```
module ModuleA::<Foo: u32>(
	x: output logic
) {
	assign x = 1;
}

module ModuleB {
    inst instA: ModuleA::<1> (
        x: _,
    );
}
``` 
fails to build with

```
Error: anonymous_identifier_usage (link)

  × Anonymous identifier can't be placed at here
    ╭─[<...>/src/test.veryl:9:12]
  8 │     inst instA: ModuleA::<1> (
  9 │         x: _,
    ·            ┬
    ·            ╰── Error location
 10 │     );
    ╰────
  help: 
```

If the generic parameter is removed from ModuleA, it builds fine, so i think the issue is somewhere there.

```
module ModuleA(
	x: output logic
) {
	assign x = 1;
}

module ModuleB {
    inst instA: ModuleA (
        x: _,
    );
}
```

taichi-ishitani, 2025-05-26:
I think your issue is resolved by #1651.
Can you try to use the nightly build version?

onsdagens, 2025-05-26:
Yep, seems to work now, thanks!