veryl-lang/veryl #1674: Resolving of local nested dependencies

onsdagens, 2025-05-26:
Hi!

Assuming a file structure like this:
 
```
top/
├─ Veryl.toml
├─ dep/
│  ├─ Veryl.toml
│  ├─ subdep/
│  │  ├─ Veryl.toml
```
And a (local) dependency tree such that `top` depends on `dep` which depends on `subdep`

`top` will not build if  `Veryl.toml` of `dep` specifies `subdep = {version = "0.1.0", path="./subdep/"}`.

It does however resolve if the path provided in `Veryl.toml` of `dep`  is relative to `top` (i.e. `subdep = {version = "0.1.0", path="./dep/subdep/"}`)

taichi-ishitani, 2025-07-29:
I think this issue has beenn resolved by #1700.
The latest release includes this PR.

Can you check if your issue is also resolved?

onsdagens, 2025-07-29:
Seems resolved with latest Veryl. Thanks!