veryl-lang/veryl #1736: Installation fail

hpretl, 2025-06-26:
On using a Ubuntu 24.04 machine inside a Docker VM, I can run `cargo install verylup` and it creates the executable. If I then run `verylup setup --verbose` I get the following error message:
```
/tmp > verylup setup --verbose
[DEBUG]     starting new connection: https://github.com/
Error: unknown target
```

Please let me know how to debug and fix this issue.

dalance, 2025-06-27:
I added information for debug.
Could you try `cargo install --git https://github.com/veryl-lang/verylup` instead of `cargo install verylup` ?

hpretl, 2025-06-27:
Ok, now it seems clear what is going wrong. Here is the same procedure as above, but on an `x86_64` machine:
```
/tmp > verylup setup --verbose
[DEBUG]     starting new connection: https://github.com/
[INFO ]  downloading toolchain: latest
[DEBUG]     starting new connection: https://github.com/
[DEBUG]     starting new connection: https://objects.githubusercontent.com/
[INFO ]   installing toolchain: latest
[INFO ]     creating hardlink: veryl
[INFO ]     creating hardlink: veryl-ls
```

hpretl, 2025-06-27:
Do you have plans to support `aarch64`?

hpretl, 2025-06-27:
Sounds good, we will then postpone Veryl for the next release.

closed 2025-07-08 by hpretl