veryl-lang/veryl #1805: Support `bind` declaration

taichi-ishitani, 2025-07-24:
We have to add te correct prefix/suffix to clock/reset signals in a `bind` declaration by hand.
Example:

https://github.com/taichi-ishitani/rice/blob/4a93c674ed7479e3beff3e06df28e1f1f306f543/tb/core/tb.sv#L39
(Module specified by `TB_RICE_CORE_ENV_INST_CHECKER` macoro is generated from Veryl code.)

Those prefix/suffix can be added automatically if Veryl supports `bind` declaration.

taichi-ishitani, 2025-09-01:
SystemVerilog allows to bind a module to a specific module instance.
To support this, a lot of changes will be required on namespace resolution.

Therefore, for now, bind to a module is only supported.

dalance, 2025-09-04:
About `bind` syntax, I have a concern.
Concatenated two `:`s in `bind` declaration may be inconsistent because `:` is normally used like `instance: Type`.

```systemverilog
bind ModuleA: u: ModuleB (
  i_clk: i_clk,
  i_rst: i_rst,
  i_a  : a    ,
  o_b  : b    ,
);
```

So how about the following ideas?

```systemverilog
bind ModuleA u: ModuleB (

bind ModuleA for u: ModuleB (

bind ModuleA to u: ModuleB (
 ```

taichi-ishitani, 2025-09-04:
Use `<-` symbol instead of keyword.

```systemverilog
bind ModuleA <- u: ModuleB;
```

dalance, 2025-09-04:
`<-` looks good!

closed 2025-09-04 by dalance