veryl-lang/veryl #1870: Whether to support the use of alwsys block

Tim-Dong-CN, 2025-08-22:
Whether to support the use of alwsys block
Such as
``` systemverilog
begin: blockB // block name after the begin or fork
...
end: blockB 
```

taichi-ishitani, 2025-08-22:
What is purpose of named always blocks?

Tim-Dong-CN, 2025-08-25:
Both sequential and parallel blocks can be named by adding : name_of_block after the keywords begin
or fork. A named block creates a new hierarchy scope. The naming of blocks serves the following
purposes:
- It allows local variables, parameters, and named events to be referenced hierarchically, using the
block name.
- It allows the block to be referenced in statements such as the disable statement

An unnamed block creates a new hierarchy scope only if it directly contains a block item declaration, such as a variable declaration or a type declaration. This hierarchy scope is unnamed and the items declared in it cannot be hierarchically referenced.

All variables shall be static; that is, a unique location exists for all variables, and leaving or entering blocks shall not affect the values stored in them. 

The block names give a means of uniquely identifying all variables at any simulation time.

A matching block name may be specified after the block end, join, join_any, or join_none keyword, preceded by a colon. This can help document which end or join, join_any, or join_none is associated with which begin or fork when there are nested blocks.